RoboJackets/robocup-firmware-old #17: Dribbler won't start again after it stalls out

jgkamat, 2016-08-29:
Migrated from RoboJackets/robocup-software#708

When dribbling a ball, sometimes the dribbler gets stuck and won't start again. Spinning the dribbler by hand can get it going again though.

jgkamat, 2016-08-29:
I'm not sure, I'll ask someone in firmare. When I checked they said it may be a issue in firmware but not likely. This is more of an issue just to make sure it gets resolved (we don't forget about it) and to track it.

petersonev, 2016-08-29:
We determined it was due to the SPI pins (MOSI and MISO) being accidentally switched on the PCB going into the driver chip for the dribbler. @jjones646 tested this and appears to fix the problem.

jgkamat, 2016-08-29:
:+1:, closing this unless it comes back up again.

closed 2016-08-29 by jgkamat