RoboJackets/robocup-firmware-old #5: Setup MBED OTA install

jgkamat, 2016-08-29:
Migrated from RoboJackets/robocup-software#154

guyfleeman, 2018-05-24:
@petersonev does this work as per your embedded final project?

petersonev, 2018-05-24:
I dunno if "works" is the right word but it does upload files to the mbed. The main issue is you can't change file timestamps or rename stuff on the mbed so it's pretty janky. It's also pretty slow but I can adapt it with the larger 1023 byte packets. I kinda stopped trying to improve it when we decided to switch to a new microcontroller because I can actually get it working on that, the best I can do on the mbed is get a hacky solution that kinda works.

guyfleeman, 2018-06-03:
So are we calling this possible but infeasible? @petersonev Trying to close out irrelevant issues for firmware pre comp.

petersonev, 2018-06-03:
Ya, it's infeasible for now but when we switch to the new MCU we'll definitely want it.

guyfleeman, 2018-10-29:
Closing, as impossible on MBED architecture.

closed 2018-10-29 by guyfleeman

petersonev, 2018-10-29:
Not possible on mbed but we do want it on the new micro controller so I think this should remain open even if not possible currently.