RoboJackets/robocup-firmware-old #6: Firmware error logging to file

jgkamat, 2016-08-29:
Migrated from RoboJackets/robocup-software#316

JNeiger, 2016-09-21:
Just a quick update.
I have the basic write to file already done and completed. 

I am now trying to implement a system that will clear the file before it reaches the max file size and fills up the mbed internal flash. For some reason, any sort of file size implementation is returning seemingly random file sizes. I'm still working on it.

jgkamat, 2016-09-21:
That might be a little hard as I don't think there are any logging utils in the mbed libraries (if there are, you should probably use them). 

Have you tried this method of getting the size (this works in c: http://stackoverflow.com/questions/238603/how-can-i-get-a-files-size-in-c)

JNeiger, 2016-09-21:
I have tried that in the past, but I did not notice the fine print in the mbed file system page saying...

`fseek is not supported for files opened for writing ("w")`

If you try to do it while writing, it completely destroys the mbed flash memory and you have to reformat it.

Hopefully, opening it with `"a"` works with `fseek` and I can get around that restriction. I'll be trying that route later this week.

JNeiger, 2016-10-18:
With the incompatibilities between the file logger and the radio (Detailed in #26), is this still worth the added complexity to include?

The current implementation will only allow file logging while the radio is disabled.

guyfleeman, 2018-10-29:
This was replaced by flexible slots in the radio return packet, and is a another example of mbed incompatibility/immaturity. Closing.

closed 2018-10-29 by guyfleeman