RoboJackets/robocup-firmware-old #21: Fix ninja build warning about git_version.cpp

jgkamat, 2016-08-31:
This gets printed out at every build and is a little annoying.  There's probably something wrong in our CMake files.

~~~{.sh}
ninja: warning: multiple rules generate common/git_version.cpp. builds involving this target will not be correct; continuing anyway [-w dupbuild=warn]
~~~

Moved from RoboJackets/robocup-software#640

jpfeltracco, 2016-08-31:
I'm a little confused, #11 says we need to switch to ninja for firmware builds and yet we're getting a ninja warning right now? I've seen references to ninja stuff in the CMakeLists.txt file too so is #11 just out of date?

jgkamat, 2016-08-31:
I'm not sure if this is even a problem in ninja, it's probably an issue with our setup of cmake. We do a few jank stuff to get the git version embedded into the output of every firmware build.

jpfeltracco, 2016-08-31:
Yeah, my previous comment was really more about #11. This is not a Ninja problem from what I can tell.

I messed around with the cmakelists a little bit and I believe these two lines are in conflict (one in the common repo and one in firmware/common2015):
1. https://github.com/RoboJackets/robocup-common/blob/ab0c8438e8600d482bc2fc64cfda70c66a86015c/CMakeLists.txt#L28
2. https://github.com/RoboJackets/robocup-firmware/blob/master/firmware/common2015/CMakeLists.txt#L95

Removing the bits about git_version.cpp in either one of them removes the error, but I haven't looked in depth enough to know what the correct thing to do here is yet.

jgkamat, 2016-08-31:
That git_version.cpp build target is used for implanting sha version info in the firmware:
https://github.com/RoboJackets/robocup-software/pull/221
(This was a while ago, when I didn't know any cmake whatsoever).

If removing the build target fixes it, try making a commit and trying again (it might break with the new sha), and then once we have access to the mbeds, we can test out that info command again. Back then, it didn't seem to work for me, so thats why both of those targets are there.

jpfeltracco, 2016-08-31:
I added a dummy commit and it didn't break the build. I'm not entirely sure what that is supposed to test, but I think I may be misunderstanding something.

I have my own mbed at my dorm so tonight I can go ahead and test the info command to see if it returns the most recent git hash of robocup-firmware.

jgkamat, 2016-10-26:
Yah, I think that this is finally starting to make sense to me. We have git_version.cpp in the common Cmake, but we also have it in the firmware cmake, hence the error. I'm converting this to an intro issue.

Removing the git_version.cpp part from this line should fix the issue: https://github.com/RoboJackets/robocup-firmware/blob/master/firmware/common2015/CMakeLists.txt#L99

Let us know if that dosen't fix it!

JNeiger, 2019-08-01:
We dont have common anymore so I'm just going to close this?

closed 2019-08-01 by JNeiger