RoboJackets/robocup-firmware-old #25: Setup build system to build hw-tests.

jpfeltracco, 2016-09-20:
We currently have a good amount of tests under [hw-tests](https://github.com/RoboJackets/robocup-firmware/tree/master/firmware/robot2015/hw-test).

From what I can tell, these are mainly just "main.cpp" files that were running on the MBED and got renamed.

- Looking into editing makefile, adding dependency on tests.

jpfeltracco, 2016-09-21:
It looks like we already had that functionality at some point based on the following: https://github.com/RoboJackets/robocup-firmware/blob/master/makefile#L57.

jpfeltracco, 2016-09-21:
Confirmed, `make robot2015-test-attiny` works.

jjones646, 2016-09-21:
~~`wait(0.3);`~~

```
#include "rtos.h"
...
Thread::wait(300);
```

jpfeltracco, 2016-09-27:
Okay, you were right @jjones646

The hardware test I was working on didn't have any wait calls so I didn't convert anything over to `Thread::wait(<time>)` and therefore this didn't fix my issues.

This function **must** be called within our hardware tests or else the above linker error will occur.

So is this just a requirement of RTOS threads or is this an issue in the build system? It shouldn't be much of a problem because just about any program should be using these waits at some point.

jpfeltracco, 2016-10-10:
So hw-tests build fine. We don't link properly when not referring to a function in Thread, but this is a different issue, and one that shouldn't affect our codebase, therefore I'm closing this.

closed 2016-10-10 by jpfeltracco