RoboJackets/robocup-firmware-old #31: Robot powers all motors when reset and on ID 0

petersonev, 2017-01-16:
Sometimes when reset robot with ID 0 will power up all of the motors at full speed and not stop until powered down. Initially suspected to be inaccurately picking up packets meant for the base station but does not stop when radios turned off so there is some other problem as well.

jpfeltracco, 2017-09-18:
Is this actually still an issue? Because this sounds pretty bad

petersonev, 2017-09-18:
I'm like 80% sure I fixed this but I'll check and test to verify

guyfleeman, 2018-06-03:
@petersonev was this fixed?

petersonev, 2018-06-03:
That's one of those things I'm 99% sure I fixed at some point but it's important enough I feel like we need to verify it somehow. Also there was an issue where the last couple bytes of the radio packet got overwritten with the checksum of the decawave so that's unlikely the cause of this but can cause issues similar to this

petersonev, 2018-06-03:
Oh wait I remember now, this was because of some filtering that I didn't have quite right on the decawaves that I fixed. So this can probably be closed as long as no one remembers experiencing anything like this

guyfleeman, 2018-06-03:
@petersonev your call can always be reopened