RoboJackets/robocup-firmware-old #50: Fix Corruption at End of RTP Bot to Soccer Packet

jpfeltracco, 2018-04-18:


guyfleeman, 2018-06-03:
@jpfeltracco got any ideas what's going on?

guyfleeman, 2018-06-04:
@ashaw596 thanks for the info
@petersonev any updates?

JNeiger, 2018-06-04:
I think this is still happening. When @jpfeltracco was implementing the encoder data forwarding to soccer, we had to add a [buffer](https://github.com/robojackets/robocup-fshare/compare/4defb667d02aeed1475be6b8db9b20420f7e4bd1...fc3b0524781967bd62dc2f4ee18fec6a4534b480#diff-f9f70a1c395cf8887effe02d2a2b9f80R112) at the end so the last encoder wouldn't get corrupted.

guyfleeman, 2018-06-04:
@JNeiger so wait, you understand the issue?

ashaw596, 2018-06-04:
Just looking through the code base a bit so not sure.
Could it be this line? Not sure whats its suppose to do but it was added a year ago.
The comment says its suppose to delete the last two elements. (Idk why though)
It actually deletes the 3rd and second to last elements and leaves the last elements.

` // remove the last 2 elements`
` buf.erase(buf.end() - 3, buf.end() - 1);`

https://github.com/RoboJackets/robocup-firmware/blob/d51a6d3b0b03ffa08991147a7b44068ad39fd75f/lib/drivers/decawave/Decawave.cpp#L139

JNeiger, 2018-06-04:
@guyfleeman I'm not sure what is causing it. I just know that when we were doing the encoder forwarding, this issue popped up.

At the very least, if you remove that buffer value in the packet, it should be really easy to test if the problem still persists.

guyfleeman, 2018-06-04:
@jneiger what branch is current for motion control data where we can try to reproduce?

guyfleeman, 2018-06-04:
@ashaw596 good catch. I'll try to reproduce soon and see if modifying that line fixes it

JNeiger, 2018-06-04:
@guyfleeman I don't think we made a specific branch to make testing the issue easy. The encoder stuff should already be on master. "Theoretically" just removing the buffer item from the packet should be enough to easily notice the problem when you print the encoder values to the console. 

These are the relevant PRs for this
https://github.com/RoboJackets/robocup-firmware/pull/55
https://github.com/RoboJackets/robocup-software/pull/1129
https://github.com/RoboJackets/robocup-fshare/pull/1

jpfeltracco, 2018-06-04:
Yeah, just remove the dummy packet field called "buffer" or something and look at encoder 4 in the tree under the RX packet item. It won't make sense.

PR #59 (2018-06-08): Fix Corruption of End of packet
Fix erase iterator to actually erase the last two bytes of the payload instead of the 3rd and 2nd to last.
Fixes #50
closed 2018-06-08 by jpfeltracco