RoboJackets/robocup-firmware-old #54: Verilog Improvements

guyfleeman, 2018-05-24:
Okay, so the Verilog is in rough shape for a few reasons. The biggest is that it was clearly written as Jon was learning (<3 you Jon, this is why we have working robots). 

Housekeeping/Janitorial
- [ ] rewrite in System Verilog style
- [ ] replace in-line ternary assignments with always_comb blocks
- [ ] replace localparam states with enums
- [ ] move module params/wires/defs into header
- [ ] cleanup generated localparams
- [ ] remove magic numbers
- [ ] move Phase_Driver.v into BLDC directory
- [ ] clarify comments in BLDC_Driver.v about moving data across async domains
- [ ] change file extensions to .sv
- [ ] rename clk divide to meet naming conventions

Improvements
- [ ] more thoroughly (exhaustively) verify hall states, read Maxon docs to verify gray enc
- [ ] verify hall count against duty cycle and expected increment
- [ ] add more types of motor/hall/encoder error
- [ ] add better error type reporting and resetting to radio protocol
- [ ] stop using the preprocessor for feature selection. If it works it should be a module param
- [ ] SPI Slave CDC mechanism needs to be impl tested
- [ ] SPI Master and Slave should probably be replaced with Xilinx IP

Testing
- [ ] more test benches
- [ ] even more test benches
- [ ] probably a test bench framework

Documentation
- [ ] BLDC high level
- [ ] BLDC lower level
- [ ] Fixing _s postfix register names in BLDC_Driver.v
- [ ] SPI Slave CDC documentation or switch to Xil IP

guyfleeman, 2018-05-24:
Tagging @jjones646 @petersonev on this for input. PR soon (TM)

petersonev, 2018-05-25:
Lol I think it might be time to update our chip

guyfleeman, 2018-05-25:
I specd out the lowest end Spartan 7 DK carries. Its about 4-4.5x more capable than the 3E

petersonev, 2018-05-25:
Do we know of a decent dev board we can get or should I just make a PCB we can use as a dev board for spartan 7?

guyfleeman, 2018-05-25:
I think my coworkers like the ARTY S7

guyfleeman, 2018-05-25:
But I imagine the changes are minor for us. Verify the pins have the same IO capability and rewrite the pinout file. We avoided using literally any Xilinx IP so none of that would need updating.

guyfleeman, 2018-05-25:
I'm going to open an issue on robocup-pcb for a new control board

petersonev, 2018-05-25:
Do we want to start with that or just design a prototype control board with the spartan 7? We can get it fully working in sim before we even move on to that

guyfleeman, 2018-05-25:
So the Vivado Sim is pretty good. Unfortunately Icarus is a little behind on SV. I see this as having two big things to work toward.

1. We need a new board
2. We need new Verilog features (and new toolchain)

Honestly, these can probably be done in parallel. Thoughts?

guyfleeman, 2018-05-25:
I also wanna get @matthewawhite  tagged in here.

guyfleeman, 2018-06-02:
Merging BLDC_Motor and BLDC_Motor_No_Encoder because this should be generated by a parameter

guyfleeman, 2018-06-02:
Merging all Hall_Effect_Sensor_* files because this should be generated by a parameter

guyfleeman, 2018-06-02:
Combine BLDC_Hall_Counter.v and BLDC_Encoder_Counter.v because this should be selected with a parameter

jjones646, 2018-06-02:
Ok, let's see if I can remember some of this:

- `BLDC_Motor_No_Encoder.v` was created because there wasn't enough logic for an unused encoder on the Spartan 3. Same thing goes for why `Hall_Effect_Sensor_*.v` files are preprocessed and why the `BLDC_Hall_Counter.v` (8 bits) and `BLDC_Encoder_Counter.v` (15 bits) files were separated. I imagine this had a lot to do with poor optimization by the synthesizer too. Lots of random refactoring like that had to be done to fit it onto the Spartan 3.
- The [then] deprecated ISE webpack was...trouble. Much much much trouble.
- All SPI stuff should definitely be replaced with Xilinx IP. I tried at the time and failed. Writing the SPI master and slave simply took less time than figuring out how to integrate IP into a headless build process. I have high hopes that @guyfleeman knows more about this and knows what to do.
- When the hall states aren't correct, the motor doesn't turn (most likely) or sounds really janky when turning (less likely). We ran into both of those cases multiple times. Currently, the expected current at no load and max speed of the motor matches what's expected on the datasheet to a tack. The hall states are 100% correct.
- Did I mention that ISE webpack is bad and causes trouble?
- Simulation is great for writing this stuff, but always do full hardware synthesis before committing. There's some verilog stuff that just can't be synthesized to hardware - or at least ISE couldn't...
- When moving to a new chip and toolchain, triple check the pinouts. And then triple check them again. And then double check them. Now get a good nights rest, and do the same thing the next day. The pinouts are very easy to screw up. _Especially_ when the pcb and verilog and being developed in parallel. It's easy to accidently configure the fpga for a v1 pcb that's physically soldered to a v2 pcb...

closed 2022-05-14 by Arvind-Srinivasan