RoboJackets/robocup-firmware-old #100: Broken Links on robocup-firmware README

dev4ughn, 2020-03-26:
Almost all, if not all, links under the Getting Started heading appear to be broken.
https://github.com/RoboJackets/robocup-firmware/blob/master/README.md#getting-started
More specifically the links for the following are missing:
- [ ] Getting Started Pages
- [ ] Overview Architecture of RoboCup Firmware
- [ ] Architecture of kicker
- [ ] Architecture of fpga

JNeiger, 2020-03-26:
It looks like the folder name was changed. From `docs` to `doc`. It should be as simple as moving everything into the docs folder, but that looks super nasty right now due to doxygen? I'll just change the links for right now, but in the future we should do it the right way and move everything into the single `docs` folder.

CollinAvidano, 2020-04-01:
Thank you Joe yes this is it. It was not changed from docs to doc. Doc is the base with the images and docs is the target directory of the make target I added a while back for documentation. Unfortunately to be used by github pages its required to either start the index.html at the root or in /docs of the repo.

PR #101 (2020-04-01): Fixed links
`docs` is super filled with stuff right now, so to keep it nice I'm just changing the links instead of moving the files from `doc`

Fixes #100 (Ayy)

@dev4ughn Can you make sure this fixes it?
closed 2020-04-01 by CollinAvidano